tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #2: Not connecting to WiFi

superman1950s, 2021-04-19:
I've flashed usb_share_v2_3_0.img.zip onto a 16GB scandisk SD card with Win32DiskImager.
I've added the wpa_supplicant.conf file with notepad++ and nothing is working. I've reformated and reflashed the img to the SD card a few times. I've even added an ssh file. When plugging the sd back into the PC, the ssh file is out of the boot, and the wpa conf is still there. In my router page, I do not see the pi zero being added to the list of connected device.  Has this been an issue, and is there a way to fix this?

superman1950s, 2021-04-19:
[wpa_supplicant.conf.txt](https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/files/6339370/wpa_supplicant.conf.txt)

tds2021, 2021-04-19:
When you added that file to the boot partition on your SD card, did it have the .txt extension?  It should not.

closed 2021-04-20 by superman1950s