tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #3: Camera crashes Pi

JobyB, 2021-04-19:
Hi,

I've spent most of the day playing with this :) for some reason turning on the camera crashes my pi zero w It's rock solid if I don't turn it on. If it's on the pi will crash after about 10 seconds of video. I've connected the same camera to my Pi 4 and its runs for hours (I've not seen it crash).

On googling there seems to be a number of threads, most discuss overclocking the pi and overvolting the power to make it more stable but neither of those have worked for me.

Have you managed to get a stable video feed with a pi zero? If so how :)

tds2021, 2021-04-19:
I haven’t had any issue with a camera crashing the Zero. I wonder if anyone else has seen that.

JobyB, 2021-04-22:
Very odd, I purchased a second brand new pi zero w, and same thing, runs fine until I turn the camera on, then crashes the OS within 10 seconds. I know its not your code, I installed a clean OS and just turned on the camera with a simple python script from the web, that crashed it too...

I'll switch to a Pi 4 and see if thats any better. Been running the software on another pi zero without the camera and loving it!!! Nice work!!

tds2021, 2021-04-23:
can you post an img copy of your SD Card that I could download and try here? Because I cannot reproduce the issue, I can't make any modifications.

JobyB, 2021-04-23:
I can, but it's a direct write of your image, I write the img to the card, stick it in the pi, boot up, enable camera and boom it starts crashing, if I don't enable the camera it runs fine.

As I said, its not your software, it does it the same with an official OS image and enabling the camera, I started to think that it was the camera as it did it on both my Pi Zeros, but same camera is running fine on a pi 4, so it must be that the pi zeros just can't handle the camera

tds2021, 2021-04-23:
Is it a V2 camera?  This si the one I have been testing with.

https://www.pishop.us/product/raspberry-pi-camera-module-v2/

JobyB, 2021-04-23:
No I went cheap: Aokin for Raspberry Pi Camera Module, 5MP 1080p with OV5647 Sensor Video Camera

https://www.amazon.com/gp/product/B07RXKZ1KN/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1

Guess its my own fault for not buying orriginal.

tds2021, 2021-04-23:
I wonder if that is part fo your issue.  You have to set specific modules in teh code when installing the camera software.  It coudl be that your modules are missing.

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

tds2021, 2021-04-24:
Based on your confirmation that it is not my image, and more likely the camera module, I am closing this issue.

closed 2021-04-24 by tds2021

Kalafkalasch, 2021-05-20:
I would like to reopen this issue and also provide additional data.

I am using a wide angle module which should be genuine https://www.berrybase.de/raspberry-pi/raspberry-pi-computer/kameras/kamera-mit-120-176-weitwinkeloptik-f-252-r-raspberry-pi-zero-15cm?number=RPIZ-CAM-WB

I also have crashes when using the camera, but not after few seconds, more like 10 minutes to 1/2 hour.

It can be seen that short before the crash the ping to the module gets high and the camera feed starts to lag behind.

If the camera is closed, the ping  will go back to normal and the camera can be reopened.

On an own install I was able to get the camera stream running stable by reducing the frames.

tds2021, 2021-05-20:
In version 2.4.2, the camera is reduced to 640x480 and 5 frames per second. you could modify the files in ~/usb_share/scripts/camera_templates/ if you want to try different settings.  IF you change them, disable the camera and re-enable it so the corrected file get loaded in /usr/local/share/start_camera_stream.sh

Kalafkalasch, 2021-05-21:
Thanks a lot.

I downloaded the new beta version and my raspberry is now being stable. I even was able to remove the "force_turbo" setting.
*The camera resolution is set to 320x240.

For my camera I set it to 1944x1296 to get more details (Rotate 90°), full width half height.

tds2021, 2021-05-21:
Sounds good. I don’t expect to touch those files in the future, but you now know what to change.  I’m glad it’s working for you.

Kalafkalasch, 2021-05-22:
Small correction on my side:
It went much better with the new version, but at a certain point it would still crash.
It is now running 2days after I kept the option force_turbo at 0 but increased the option over_voltage to 1.

Great Project, thanks for your share