tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #4: Known Issue: Starting a print job may load the wrong file.

tds2021, 2021-04-20:
By sorting the file list on the Mono X continue page, a bug appears to have been introduced. The printer may load the incorrect file. I’ll fix this over the weekend.

JobyB, 2021-04-23:
This hit me today :(

lollokara, 2021-04-23:
Currently a fix is to delete all the files but the one you need to print. I will try to have a look at the code and see if I can get some sort of fix

tds2021, 2021-04-23:
The issue has been resolved and a patch is available.  Install instructions are here:

https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.3.1-patch

I will create a new img file as well, but that take a few hours.

I belive this is resolved.

JobyB, 2021-04-23:
Great thanks, just downloaded it and installed, love the other little changes like the fav.icon I'll test out the fix a bit later when I'm not printing something :)

tds2021, 2021-04-24:
2.3.1 patch resolved this.  Confirmed by two testers

closed 2021-04-24 by tds2021