tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #6: Feature Request: Time laps feature

JobyB, 2021-04-23:
Hi,

Would be really nice if you could include a time laps feature to be able to create time lapses of builds.

Thanks

tds2021, 2021-04-23:
I looked at this, but the Zero W could not handle the assembly for the images into a video.  At best, I can create a directory full of all the sill images (4/sec) and zip them when the print is done. You would have to download them and create the video on your own.

JobyB, 2021-04-23:
Ahh ok, the zip of pics would be good at least, or maybe a pi 4 feature only? I'm switching to a Pi 4 due to my camera issues

tds2021, 2021-05-11:
My testing points to the Pi Zero not being capable of handling this.  The existing video stream is already taxing the CPU.  I will not be trying to add timelapse video capabilites.

closed 2021-05-11 by tds2021