tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #7: automatic Update runs into 404

PrCRTL, 2021-04-23:
tried to Update to 2.3.1 by picking the Upgrade button... runs into "404 not found" Page

tds2021, 2021-04-23:
Please see the notes on teh release page for 2.3.1. There is a fix for that as well in 2.3.1, but you need to run the command from ssh.

https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.3.1-patch <https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.3.1-patch>

sudo upgrade_usb_share

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

PrCRTL, 2021-04-23:
yeah, sorry was a bit too unpatient have just seen it...
Thanks for this great Tool !!

closed 2021-04-23 by PrCRTL