tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #8: U disk not detected

WheresWaldo, 2021-04-24:
Up and running on a rPi Zero W using a 32GB Samsung Evo card for the image. USB Share is 16GB, running 2.3.1 manually updated per instructions from 2.3.0
[image]
[image]
[image]

Files can be copied from the windows share, as seen in the Summary screen, but do not appear in the Mono X screen. In addition when checking with the Anycubic Android app when trying to view the file list the message "U disk not detected" appears in the middle of the screen.

Files print normally off a standard 32GB USB Flash drive. Printer is on firmware 3.4.7. Resetting USB or rebooting rPi from the Power/Control screen does not fix the issue.

tds2021, 2021-04-24:
Couple of questions:

Are you certain you are plugging the printer into the correct miso USB port?  Both will power the Pi Zero, but only one will pass data to the printer.  It has to be connected to the port closest to the middle of the Raspberry Pi.  If you look really close on the circuit board, you will also see they marked it “USB."  You do not want to use the port marked “PWR.”

Additionally, was the printer already powered on when you connected the Zero?  90% of the time this works fine.  I have probably done this 100 times when testing.  However, the printer is finicky and behaves strange from time to time. Every now and then, including yesterday with my RPi 4, when I connect a Pi to the Mono X it will not see the USB interface.  It does this with regular USB drives as well.  The solution has been to power of the printer, what a few seconds, and then power it one with the USB in place. It won’t help you, but when this happens, I have noticed the printer is sending an ERROR1 code for the file list.  Also, the list will be completely empty.

I would suggest making sure you are connected to the correct port on teh Zero, and then power cycling the printer with the zero already connected. 

Finally, to clarify, the file on the Summary tab is long at what you have really loaded on the USB drive. It can be your pwmx files, text files, whatever.  This is a generic solution for any use.  What you see on the MonoX page is what the printer has loaded.  Again, this won’t help you, but I think is interesting.  You might load a file named cool_figure_I made.pwmx.  Taht is what is on the local storage for the RPi.  What comes back from the printer is something like cool_figure_I made.pwmx/42.pwmx.  Every file is given a new name that starts with a number.  The reason the list on the MonoX page is waiting for the info to come back from the printer, and not looking locally is that the ##.pwmx file is the name that matters.  That is what has to be sent to the printer to start a print job.  The name you gave the file is not used for anything other than display. I’m sure there is a reason they designed it this way, maybe internationalization. Etiehr way, you might be stuck in the printer being able to assign those extra names; it might also be what makes the power cycle required from time to time.

There is always a 30sec delay from when you load a file to the Raspberry Pi until it appears on the MonoX page. This is the delay for the USB to be unplugged and reconnected. That way you can upload multiple file and not have the printer disconnecting for each one. 

Please let me know fi this info helps you get up and running.

Troy

WheresWaldo, 2021-04-24:
A couple of answers:

Yes I am certain it is plugged into the USB not the PWR only port
[image]

Yes printer was already powered on when I plugged in the rPi.  Powering off the printer and plugging in the rPi then powering on the printer does not affect the issue. It still does not recognize the rPi as a USB Flash Drive. I know the Mono X works because I can plug in any one of several USB sticks I have (Sandisk, Toshiba, no name Chinese) and so far all work and I have printed from them all.

I assumed the files on the summary tab was local to the file system and that they would need to be 'recognized' by the Mono X before being seen on the Mono X page. I did not know they are physically moved/renamed. Nevertheless, this is not the issue as the printer never gets to this point since it does not believe there is a USB Flash Drive attached.

I waited hours, not just the 30 seconds mentioned in your write up. I can try multiple files at a time, to see i that changes anything.

If there is more I can do to troubleshoot, just ask.

tds2021, 2021-04-24:
Not really;  let me noodle on it.  

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

WheresWaldo, 2021-04-24:
Here is what the Anycubic 3D android app sees:
[image]
[image]

This leads me to believe that the Mono X simply doesn't recognize the rPi as a USB drive at all.

WheresWaldo, 2021-04-24:
Just for grins here is the output power of the Mono X USB port:

[image]

tds2021, 2021-04-24:
Do you have a different cable to try?  

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

tds2021, 2021-04-24:
Have you tried plugging it into a USB port on your computer?  IT shovel appear as a drive there as well..

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

WheresWaldo, 2021-04-24:
Don't let anyone tell you that you don't know your stuff! 🤣 🥇 

I had two cables I was using and neither worked. I dug around a lot, seems Micro USB was not something I needed a lot of cables for. I found a short 3" cable and, you guessed it, it works! 

[image]
[image]

tds2021, 2021-04-24:
Great!  

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

WheresWaldo, 2021-04-24:
I am closing this issue.

closed 2021-04-24 by WheresWaldo