tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #11: Feature: Auto-shutdown after print finish

Hesi-Re, 2021-04-25:


PrCRTL, 2021-05-16:
only useful if is an switchable option

Hesi-Re, 2021-05-16:
Yes, of course

olituks, 2021-07-21:
Could I suggest to add the possibility to sent http or mqtt request before the shutdown command.
Like this a smart switch or a Node Red server can be triggered to cut the power or performing any other actions.

lollokara, 2021-07-21:
You will need to setup a delayed off, since from the request to the actual shut down there will be some time, shutting the printer and consequently the raspi will cause issues in the long run if the shutdown is not complete. A suggestion I might have is to rely on a second device or (don't know if is possible) setup a timer in a sonoff or the smart plug you have to shut down at a certain time that is now + 1minute. That is just some brain farting...

closed 2021-12-03 by tds2021