tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #12: Include command on disabling wifi power saving on the code page

WheresWaldo, 2021-04-25:
The RaspberryPi Zero has wireless power saving on by default and the usual way of turning it off is to issue the following command in an SSH shell:

`sudo iwconfig wlan0 power off`

While this works, it does not survive a reboot of the rPi. There are a lot of scripts and posts that people have written on the subject and most work based on specific configurations. Why would we want to do this,? By disabling wireless power saving we have instant access to the web interface at all times with absolutely no delay waiting on the rPi to wake up. If we add the line:

`/sbin/iwconfig wlan0 power off`

to the /etc/rc.local file by inserting that line just above the 'exit 0' it will disable power saving for the wifi module on every single reboot. If people are already using your image then let's assume they know how to edit a file in Raspbian. Or, you can include editing information in your project description.

tds2021, 2021-04-25:
Great find.  I’ll add this to a future build

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

tds2021, 2021-05-06:
this has been fixed in 2.4.0-beta.  I did not see any performance differences, but your suggested rc.local change has been included. 

https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.4.0-beta

closed 2021-05-06 by tds2021

WheresWaldo, 2021-05-06:
You won't see any performance difference in testing, it is only when wifi
goes to sleep that the startup takes measurably longer. With this added
there is no start up delay after inactivity. It is more of a convenience
feature than a performance feature.