tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #14: Feature request: Photo trigger

PrCRTL, 2021-04-25:
Timelaps is always a request but is not really working well to drive it by time and looks very "bumpy" 
What about to define a pin on the Zero which takes a picture if it gets an 3.3-5V impulse...
The simple and perfect thing is that with an simple Light sensor we are able to take an picture when LED starts curing.
So I get a series of Pictures which can easiest combined to a Video which showes a smoothest "Resinlaps" as possible.

best regards

JobyB, 2021-04-27:
I like the idea, but given that the LED will come on at a set interval (ignoring the base layers) why not just have the interval configurable, if you know it takes 3.9 seconds per cycle, set it to that. Having it automatic using a light sensor could be nice, but its a lot more work to set up, and it could also be fooled by an external light source.

PrCRTL, 2021-04-27:
The Answer is simple....On an monox I never was able to calculate the exact Cycle time .. there are some hidden things going on like Transition layers and everything goes out of sync... tried this several times with different cameras an tools.Some cameras support external trigger but they are in an upper price range...or they only support fixed rates like my GoPro... or iphone.So my hope was to win you to get this 😉Best regards

tds2021, 2021-05-06:
this is outside of the scope of this project.  Other than the Pi camera, I am not integrating other hardware.

closed 2021-05-06 by tds2021

PrCRTL, 2021-05-06:
may be my explanation did lead into the wrong direction... I dont suggest to integrate new hardware, I only asked for an trigger pin which tells the Pi camera to take a picture in every event when a pin gets High or low... Time triggered is simply poor...