tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #15: Feature Request:Support Photon Mono SE

NoOneAtHome, 2021-04-26:
Would it be possible to support the Photon Mono SE.  They both use the same Mobile app to connect to the printer.  I am guessing their protocols would be very similar.  The main difference being that the Photon Mono SE uses .pwms files vs .pwmx

tds2021, 2021-04-26:
I do not have an SE to reverse engines the API, and AnyCubic does not publish any details on this for either printer.  I’m not sure how I would figure out what the commends are that need to be sent to the printer. Have you tried using the Mono X config?  If it works, let me know. 

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

NoOneAtHome, 2021-04-26:
I have tried and it "kind of" works.  After enabling Mono X Support with the Mono SE IP address the mono_x.php page shows that it is connected, but does not list any of the files.  When I start a print from the printer I do see the status on the mono_x.php page.   I think the only difference between the two printers, protocol wise, is that the mono SE uses , .pwms files and the mono uses .pwmx files

tds2021, 2021-04-26:
Ok.  I’ll see about allowing the pwms file types as well.

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

tds2021, 2021-05-06:
I have added settings for selecting the Photon Mono SE to the 2.4.0-beta image. However, I don't have a way to test it.  Hopefully the changes I made work for you.  Let me know.

https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.4.0-beta

NoOneAtHome, 2021-05-08:
The files now show up, however there is no action on the Start Print button.  I looked at the anycubic_mono.php file,  around line 391 and there is an if statement only looking for .pwmx then adding the listener.  I added an elseif to get it to work but I am sure there is a better way to tokenize it 

Code I added around line 391 in anycubic_mono.php
          if (strpos(strtolower($fileNames[0]), '.pwmx') !== false)
                {
                    echo '
                    function startPrinting' . $key . '() {
                    action = \'print,' . $fileNames[1] . '\';
                    monoXControl(action);
                    }
                    document.getElementById("startPrinting' . $key . '").addEventListener("click", startPrinting' . $key . ');
                    ';
                }

tds2021, 2021-05-08:
Did you re-image your SD card with the new Img file(2.4.1)? That block was changed to:

<?php
    foreach ($fileList as $key => $val) {
        $fileNames = explode("/",$val);
        
        echo '<tr>
            <td>', $fileNames[0], '</td>
            <td style="text-align:center;">[image]</td>
            </tr>
            ';
    }

?>

The file type restriction was completely removed.

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

tds2021, 2021-05-08:
Wait, I know where you mean.. It’s not the list being created, it’s the dynamic functions being added. Place the attached file in /home/pi/usb_share/html/root.  That should fix this for you… I hope.

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

NoOneAtHome, 2021-05-08:
Thanks but i do not see an attachment to your post

NoOneAtHome, 2021-05-08:
It works!  Thanks for the quick responses and great work on the project.

tds2021, 2021-05-11:
2.4.2 is not available. It includes the code I sent you over the weekend.  Unless I hear otherwise, I'm going to assume this issue is closed.

closed 2021-05-11 by tds2021