tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #16: Mono X don't see files

brumouta, 2021-04-27:
Hello, 
I have a brand new Raspberry pi zero W with version 2.3.1 installed in a brand new Mono X (firmware version 3.4.8).
My problem is that the printer can't see any files from the pi zero, even after manually unplugging and plugging the drive.
My USB cable is on USB entrance (not the PWR IN), and I see my files in the summary page but not the printer page or directly on the printer.

Thanks

WheresWaldo, 2021-04-27:
@brumouta 
I had the same issue, it is likely the USB cable you are using. Not all USB cables are fully populated. I tried six different cables from different sources, even some new ones purchased this week from Amazon. Here is one that I have tested and actually works, [https://smile.amazon.com/gp/product/B01LYYT3N3/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1](url)

tds2021, 2021-04-27:
Try different cables. There have been a few other people with similar issues and it was the cable. It seems like some are better than others. You can also connect the RPi Zero to your computer to validate. 

Sent while walking into stuff

Troy

brumouta, 2021-04-27:
It really was the cable, thanks @WheresWaldo and @tds2021

closed 2021-04-27 by brumouta