tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #18: Errors accessing share in Win10 - Mono X

WheresWaldo, 2021-05-06:
Version 2.4.0 Beta

Everything installed properly and the web interface almost works. The steps to create my USB drive were (short list)

1. flash 
2. insert in rPi Zero W
3. configure from web interface

Dragging and dropping a file into the web interface and waiting or resetting the USB appears to work fine.
This is on the same rPiZW with the same MicroSD with the same cable that worked perfectly in 2.3.1

The issue is the Samba share. If I try to map a network drive I can see the rPiZW and even see the share name. In my particular case it is located at \\192.168.1.38 and the share shows up as \usb. I can then connect to \\192.168.1.38\usb and Windows appears to map the drive. But, when you try to access the share from Windows Explorer you get the following dialog
[image]
If you try to diagnose the issue you get the following error dialog
[image]
Followed by this
[image]

I will try reflashing later today, but if that doesn't work I will need to go back to 2.3.1

WheresWaldo, 2021-05-06:
I don't know if you have enabled SSH by default, but I did add the blank ssh file before the first boot up, so I have SSH access to the rPi if there is anything you need to see directly on the rPiZW. Just ask and I can provide additional screenshots.

tds2021, 2021-05-06:
Run these two commands.  I’ll fix this in 2.4.1, but for now this will get you working.

sudo sed -i 's|/home/pi/USB_Share/upload|/home/pi/usb_share/upload|g' /etc/samba/smb.conf 
sudo service smbd restart

Troy Smith
***@***.***

“There are two kinds of people in the world: Those who divide everybody into two kinds of people, and those who don’t.” — Robert Benchley, 1920

WheresWaldo, 2021-05-06:
That works and I will close this with the fix coming in 2.4.1

closed 2021-05-06 by WheresWaldo