tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #21: Upgrade to version 2.4.2 BETA not working

Hesi-Re, 2021-05-15:
The button to upgrade to version 2.4.2 BETA does nothing. If I click in this button I see a message "upgrade in process" but after it I continue with version 2.4.0 installed. I tried to reboot but without success

tds2021, 2021-05-15:
What version are you on? The text on gethub stated that you would need to re image to jump to the new code. It could not be an I place upgrade. 

Sent while walking into stuff

Troy

closed 2021-05-15 by tds2021

Hesi-Re, 2021-05-15:
You are right. I thought I could upgrade in the webpage having version 2.4.0.

Despite this, I did the following steps and it seems to be working:

cd /tmp
wget https://raw.githubusercontent.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/main/resource_files/2_4_2_resources.zip
unzip 2_4_2_resources.zip 
rm 2_4_2_resources.zip 

2_4_2_resources/upgrade.sh

rm -rf 2_4_2_resources
rm -f upgrade_usb_share.sh