tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #23: Important Question

PrCRTL, 2021-05-16:
because of this I'm afraid of printing issues on the Mono:

"After updates on the network-share, the Pi waits 30 seconds and then automatically disconnects (unplugs) and reconnects the USB interface so changes appear on the target device"

What happens if the mono is transferring data and this happens just in that Moment ?
As far I know up to now, the Mono has no Cash so this may mess up prints somehow in my Understanding

I`m asking because I have Issues with periodical layershifts....
Thanks and best regards

tds2021, 2021-05-16:
Before the USB is reset, it checks if the printer stays is “stopped”. That way is it is printing, it will delay the reset until the print is done. However, if you are concerned, don’t try uploading new pwmx files while printing. Then you have nothing to worry about. You have control of when you add new content. 

Sent while walking into stuff

Troy

PrCRTL, 2021-05-17:
Thanks a lot !

 

Gerhard

closed 2021-05-17 by tds2021