tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #28: Files not showing up to print

PapaChop83, 2021-07-03:
Platform: [Rpi 4 Model B 2019 4gb](https://www.amazon.com/gp/product/B07TC2BK1X/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
USB Cable: [USB A to USB A](https://www.amazon.com/NIMASO-Double-Compatible-External-Player/dp/B08DK8RPRW/ref=sr_1_2_sspa?dchild=1&keywords=usb+a+to+usb+a&qid=1625284929&sr=8-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyNDNGV1AxTlRCUDlHJmVuY3J5cHRlZElkPUEwNDA2MDg0MTBUQ0Y5UTg0QUpETyZlbmNyeXB0ZWRBZElkPUEwOTEwODY2M0kwNU5TQ1FDRkRTQSZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=) connecting USB port on Rpi to Mono X

The Mono X is recognized by the Rpi, can navigate to USB storage, move files, the file appears under **USB Storage** in the **Summary** tab, but it does not show up on the printer or under the **Photon Mono X** tab. I've tried restarting the Rpi, switching ports on it, etc. but no luck.
[image]
[image]

tds2021, 2021-07-03:
Thank you for including the equipment list. 

The only port on a RPi 4 that supports USB OTG is the power port. You have to use a usb A to usb C cable and get the power from the printer. Keep it short. 

This is the one I used in testing. It works for me. 

https://www.amazon.com/dp/B0711C43JP/ref=cm_sw_em_r_mt_dp_4CXSX1MSE89XRZZKN9RR?_encoding=UTF8&psc=1

PapaChop83, 2021-07-03:
Thanks for the quick reply, but still no dice - it powers up fine, sees the Mono X, but won't display files on either the printer interface or through the web interface.

JobyB, 2021-07-03:
Turn the USB C plug up the other way, some are only wired for OTG one way round

tds2021, 2021-07-03:
That solved it?

Sent while walking into stuff

Troy

PapaChop83, 2021-07-03:
Huh. Learned something today - Jody was correct. Works now. Thanks!

closed 2021-12-03 by tds2021

PSmith4, 2022-01-30:
I'm having the same issue with a raspberry pi 3+ and an anycubic mono.

I have tried both USB ports (USB A to A  and USB A to micro)

Do either of these work with otg on the pi 3+?

tds2021, 2022-01-30:
The only Raspberry Pis, that I am aware of, that support OTG are the Zero W, Zero W2, and RPi 4.  All of those ha ve been tested by me and work fine.  However, some people have found that some cables are more reliable than others.  That’s why I included my exact build equipment and links to Amazon.  I know they work with my Mono X.  Unfortunately, that’s all the gear I have to test with.

T