tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #30: Feature Request: Make an option for the USB disconnect/Reconnect  to be a manual process

SarcasticShrimp, 2021-07-06:
I would be great to be able to drop files into the smb, while the printer is running, then, while at the printer, push a button (connected to the pi or via the web interface) to initiate the usb reconnection process (to allow the machine to see the uploaded files.)  Ideally this would be a configurable setting (cfg file or via web interface) so that those who like risking their active prints can continue to do so. 

Any chance this is possible?

tds2021, 2021-07-06:
I think you already have what you desire.  The script that unmount the USB interface to represent the files to the printer already checks to see if the printer is currently running.  If it is, it skips the re-load and starts a loop waiting.  The the print job does complete, it will automatically reload the USB for the printer.  It should not interrupt a print job, even if you do upload files during a print job.  Additionally, on the power/controls page there is a button where you can force the reload. 

If you want to test this, take your vat off the printer and start a small print job (maybe a RERF) as a ghost print.  While it’s running, load your extra files to the SMB share, and wait.  It’s possible that at the end of the print you have to click the reset button, but that all depends on factors like the RPi running out of resources in teh loop.  I can tell you have have loaded files a few times during a print job.  However, I suggest you test it to validate.

SarcasticShrimp, 2021-07-08:
I tested it and it did interrupt the print... I started printing a file (that I didn't care about) that was on the pi, then after a bit, I went to my computer and uploaded a new file to the pi... after about 30 seconds the printer stopped with an error "U disk has been removed!"  [Error](https://photos.app.goo.gl/BYgWEkUg2AEYzPpa6)

tds2021, 2021-07-08:
Then the Pi had a network disruption and could not get a status from the printer. It’s finniky for sure. 

Sent while walking into stuff

Troy

SarcasticShrimp, 2021-07-08:
"Status from the printer"???  How does it get a status from a Photon Mono?  I didn't think it could get status updates and control a Photon Mono.  And why would a network disruption cause any issues with it reading the status (if it can read the status via usb)?

slumcat, 2021-11-23:
> I tested it and it did interrupt the print... I started printing a file (that I didn't care about) that was on the pi, then after a bit, I went to my computer and uploaded a new file to the pi (via smb)... after about 30 seconds the printer stopped with an error "U disk has been removed!" [Error](https://photos.app.goo.gl/BYgWEkUg2AEYzPpa6)

I have had this happen to me several times recently when I've been slicing and uploading files and forgot I had a print running. I have had several dozen flawless prints with zero errors due to network issues, but the handful of times that I've uploaded a file while the printer is running I have gotten the "disk removed" error. I also have  Photon Mono, so either the print status check is not working as designed, or it is just not compatible with the Mono.

closed 2021-12-03 by tds2021