tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #31: MonoX Console empty

PrCRTL, 2021-07-06:
when start printing again around 2 weeks ago, connection to MonoX console takes much longer as before ( same Version 2.4.4 BETA) and a few days later after connected the screen is simply empty.... the only thing changed on my system was and Win10 Update.... 
Sorry but I cant be more precisely... see on my picture what I see after connected to the printer...

[image]

closed 2021-08-23 by PrCRTL

discojon, 2021-09-17:
Same thing here.  It worked exactly once when i first set it up.  But now it is completely blank as you indicate.

discojon, 2021-09-18:
Were you able to get this resolved or do i just need to reflash?

PrCRTL, 2021-09-18:
Hi, yes, it is solved. Seems it was an issue with an Win10 Update.ThanksBest regards

klim123123, 2022-01-11:
I have exactly the same problem. Just blank page after connecting... 
It worked once or two and now only blank page.
I tried Firefox, Chome, Edge, iPhone and Android. Same problem with blank page.

My printer is available using iOS app no problems.

[image]

[image]

klim123123, 2022-01-11:
It was a problem with my UniFi network. For some reason it's blocked traffic to MonoX from everywhere buy iPhone.
It would be nice to have at least error message on this screen.