tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #33: Does not mount

fiveorders, 2021-07-23:
Hello, thank you for this ready to use image. 

I can see it as a network drive on my mac but USB does not mount. It does not mount to TFT70 LCD screen as well. 

How can I make it mount like a USB stick?

fiveorders, 2021-07-23:
And also there is an upgrade button on top right but it goes no where. Thanks

[image]

tds2021, 2021-12-03:
I'm not sure what version you are using, but I would download the 2.4.4 image and start from a clean install.  As for the mounting, make sure you are using the correct per port.  Those are the only ports that support OTG.

closed 2021-12-03 by tds2021