tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #35: Feature Request - MQTT Client

jaysuk, 2021-09-15:
Include and MQTT client so status updates such as print started or print finished can be pushed to other devices

skycryer, 2021-09-16:
That would be cool so we can integrate this in Smart Home Systems or other tools.

tds2021, 2021-12-03:
I do not have the experience or equipment to test MQTT.  I will not be taking this on.

closed 2021-12-03 by tds2021

Zunhammer, 2021-12-29:
Hi,
I would also love this feature (at least reading the states to perform tasks in my home automisation) and probably could support implemeting it. Besides providing an interface for configuration, it should be quite easy to implemt it using public mqtt libraries, simply calling the publish functions with the values you already have for the web interface. Please contact me if I may have a look into your current code and assist you here.
Thanks and best regards,
Nico