tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #36: I can't find the device in the network

dade89-deve, 2021-10-03:
I created the file to connect the device to the network, writing the network name and password, checked that the file does not have a different extension, but in no way do I see the device on the network ... what could be the problem?

tds2021, 2021-10-03:
I think that would be difficult to troubleshoot remotely.  I would suggest connecting a monitor and keyboard and doing it the "old" way via the raspi-config menus.  There are step by step instructions on this page.

https://www.seeedstudio.com/blog/2021/01/25/three-methods-to-configure-raspberry-pi-wifi/

brownt12684, 2021-10-17:
Did you remember to leave the quotation marks in there? The quotation marks are needed when filling the network name and password.

Starflashbully, 2021-11-12:
Hello,
I am new here, and I tried to realize the project, but I can't see the raspi in my network, too. It's all new to me so I am also wondering why the image file is so big bit the boot folder so small at the end. For it seems that there is nothing more on the sd.  I also tried to change the country from US in DE, but that didn't help either.
Maybe someone can help me?

Thanks a lot!

tds2021, 2021-12-03:
part of the image is a disk image that is used to act as teh USB Drive.  The image is large because I already create a default "drive" for you.  You can change the size from the admin page to make it bigger or smaller based on your goals.  However, that is what explains the file size.

As for finding the device on your network, can you look at teh registered devices on your router?  That may be the best way to figure out what eh IP address is.  Also, did you configure the Pi to connect to your WiFi?  If not, you can use a monitor and keyboard to get on the device and do it now.  There are lots of examples online for doing that. IT's not unique to connecting to your printer.

closed 2021-12-03 by tds2021