tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #40: Pi zero 2 W

Zw3rg85, 2021-12-02:
Hello how can i use a pi zero 2w?
is version 2.4 down?

Thank you

tds2021, 2021-12-02:
I do not have access to a pi zero 2w so it has not been tested. However, I would probably surprised if and hardware changes in the pi would matter. As for the version 2.4.4 is the current version and has been stable. What issue are you having?

Zw3rg85, 2021-12-02:
Hello. Were i can Download 2.2.4

Mit freundlichen Grüßen
Alexander Beck

E-Mail: ***@***.***
Mobil: 0151 - 22681416

Alexander Beck EDV - DESIGN - PRINT - LASER GRAVUR
Schulweg 4, 69412. Eberbach Geschäftsführer: Alexander Beck
USt-Id-Nr.: DE316724113
www.Beck-Eberbach.de

tds2021, 2021-12-02:
https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases

Zw3rg85, 2021-12-02:
Tank you
I will test the Version with Zero w 2

Mit freundlichen Grüßen
Alexander Beck

E-Mail: ***@***.***
Mobil: 0151 - 22681416

Alexander Beck EDV - DESIGN - PRINT - LASER GRAVUR
Schulweg 4, 69412. Eberbach Geschäftsführer: Alexander Beck
USt-Id-Nr.: DE316724113
www.Beck-Eberbach.de

mnrivera210, 2021-12-02:
Hey I loaded the image onto the Pi W 2 and it will not boot. Anyway to manually install this to a clean image?

Zw3rg85, 2021-12-02:
Hello I Have Testet 2.4.4
https://github.com/tds2021/Pi-Zero-W-Smart-USB-Flash-Drive/releases/tag/2.4.4

Works Well With Pi Zero 2 W

closed 2021-12-02 by Zw3rg85

tds2021, 2021-12-02:
Not easily. However, I ordered a Zero2w this morning. When it get here I’ll see if I can make an update mage that works with the new hardware. 

Sent while walking into stuff

Troy

Zw3rg85, 2021-12-02:
[image]

tds2021, 2021-12-03:
That's great news!