tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #42: Rpi4 and lan (no wlan)

LordKerrrigan, 2021-12-12:
hello and sorry i don't speak english well.  I am using Google Translate.

I don't have wifi in my workshop.  and I tested with the wired connection and it works.
but I do not have access to the print command (stop pause)
is it possible for the rpi4 to generate a wifi that the anycubi photon mono connects to?

Exemple:
Wifi.txt anycubic
Account:rpi4,
Password:1234;),
Setting:1,

And in wpa_supplicant.conf its a same

tds2021, 2021-12-12:
Sorry.  I don’t think so.  But I’m not certain about that. My Raspberry Pi experience is limited.

T

LordKerrrigan, 2021-12-12:
me neither :( you are much better than me in this area. but I found this:
https://www.google.com/amp/s/thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/amp/

Quisi151, 2022-02-02:
Hi, thanks LordKerrrigan for the info with the wired lan connection. I'll have the same problem (no usable Wifi at the institute). So it's good to know that it also works with a lan wired Rasp4. I don't care if the start/stop works. Pushing files to the printer and having a webcam picture is fine.
Did you change any other settings to push an IP address, Subnet mask, and Gateway or did you use dhcp?