tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #43: Open the Interface via HTTPS

Zw3rg85, 2021-12-27:
Hello can i Open the Interface via https ? 
I will add the side To my Home Assistant. 
But only https is accepted in the iFrame

tds2021, 2021-12-27:
I do not know of any way to open the printer’s interface via a web browser.  If you mean my applications, you could enable https in the Apache web browser, but you’ll need to generate an SSL cert as well.  I’m sure there are instructions for that on line.  I have not done it.

T

adamoutler, 2022-02-14:
The information you need to import into Home Assistant can be obtained from this JSON API.  http://[TheIpAddress]/includes/util_anycubic.php?a=printer_status.   Replace [TheIpAddress] with your ip to your USBShare device.   You can use RESTful sensor to gather the data.  https://www.home-assistant.io/integrations/sensor.rest/

If you're trying to get the actual page in an iFrame, then you would need to put an HTTPS Reverse Proxy in front of the app.  This can be set up with Apache.
Here is an example Apache Reverse Proxy conf file.  I use this as my base when setting up reverse proxies. 
https://pastebin.adamoutler.com/F47CCdKXcQVWVNwX

adamoutler, 2022-02-14:
Aside from RESTful sensor, the development for Home Assistant has not been done. So there is no true device-level integration, but you can use that for a sensor.  A proper HA Integration would need to be created.  It's best if it's done by or with the blessing of the USBShare Author, so it gets accepted into the Home Assistant base. 

If I were to pick up this project, would you bless it, @tds2021 ?