tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #44: Support for USB Camera

CountryDevil, 2022-01-11:
I understand that the Pi is powered by the the USB port on the printer. But is there support for a USB cam on the PI? Although I have both available to use. a USB camera will give better picture quality which is what I am aiming for.

tds2021, 2022-01-11:
Using a Rapberry Pi camera module will work. Othe USB cameras will not because the printer does not provide enough power (in my tests)

Sent while walking into stuff

CountryDevil, 2022-01-14:
Thanks for the info. In that case I'd like to add a feature requests of sorts. 

So far the everything is working great, But due to power being supplied through the same USB-C port as the "drive" The Pi become limited especially when I am using a Pi 4B. 

That said, using a PI IR camera, there is not enough power from the Print USB port to power both the Pi and the camera. So my thought was to use a secondary camera set up on a different Pi and stream it to the Printer Pi. Same concept, but different IP. 

Although I could go edit the Apache files to do this, I thought it would be a nice options to manually configure the IP in you UI interface for a separate Cam stream. By default, use a PI V2 camera as you have it now, but also provide an option for a different camera IP stream. 

Just a thought for a nice little enhancement. 

Thanks