tds2021/Pi-Zero-W-Smart-USB-Flash-Drive #48: Does not work as USB Drive for Mono X 6K

adamoutler, 2022-02-12:
I am new to SLA printing and I found this project in order to replace Octoprint for my FDM printers.   I set up a basic test setup before I print something more permanent. 
Hostname:  3DPRINTER
Bonjour Name:  http://3dprinter.local  (Windows Bonjour Installer)
IP Address: 192.168.1.251
Model: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
[image]

Everything seems good to go, but the problem is the drive is not seen by the printer.  I have files available but they won't show up on the printer or in the UI.
[image]
[image]
[image]

Could this have something to do with Mono 6K? 

I'm including my dmesg as it seems to have some odd messages `g_mass_storage gadget: userspace failed to provide iSerialNumber`
https://pastebin.adamoutler.com/t1059UczyiBHytkF

adamoutler, 2022-02-12:
Additionally, all functions seem to work.  Camera, Status, SMB, and USB Mass Storage works on Desktop.  But even after using reset USB, it does not read the mass storage device.. 
[image]

sampka, 2022-02-12:
I had same issue with the mono x 6k. I found out the issue was you have to cut the red wire in your usb that is between the rasp and the printer because if the rasperberry gets any power from the printer instead of already being on and powered by wall plug it won’t recognize it as a usb drive. Although this fixes the issue your seeing I am seeing a new one with the 6k. If I run a very long print 24hours+ some point durning it a blip causes the raspberry pi to unmount from printer and make my print fail cause the printer assumed the usb drive came unplugged.

adamoutler, 2022-02-12:
That's usually the case with a serial port or an externally powered device connecting to another source of power due to the difference in voltage between the two.  In this case, the device is powered by the printer so no difference.

tds2021, 2022-02-12:
It’s hard to tell from the pictures how you have it all connected but to test things out, try plugging the PI into your computer as a “thumb drive.” It should pop up as a drive on your computer. If not, double-check which port you are using on the RPi. It looks like you might be using a USB header instead of a cable?  You might want to try a cable for troubleshooting.  I know that sounds crazy, but some people have found that not all cables are created equal. In fact, some will only pass power without the data connection.  I assume those are meant for charging only on things like phones. 

As for the question about the Mono X 6k, I have no way to answer that. I don’t have one to test with. 

T

 

Bonjour Name: http://3dprinter.local <http://3dprinter.local%C2%A0%C2%A0(Windows>   (Windows Bonjour Installer)
 <https://user-images.githubusercontent.com/527919/153724633-314c31fa-a812-4b95-ab1c-357b29800d77.png> 

 <https://user-images.githubusercontent.com/527919/153724698-1839ed6b-6743-4de3-89eb-30350ae4a3a5.png> 
 <https://user-images.githubusercontent.com/527919/153724772-ca93617c-56ca-4737-a056-be5453320e00.png> 
 <https://user-images.githubusercontent.com/527919/153724926-29aa3c60-17fc-4186-9024-0cd790225d4a.png> 

https://pastebin.adamoutler.com/t1059UczyiBHytkF

adamoutler, 2022-02-13:
It looks like you need to use one of these with the MonoX 6k.  https://www.amazon.com/gp/product/B07PV8ZN1X/

The Pi Zero W + camera slightly exceeds USB specifications and the port shuts down, until the unit is rebooted.

The expected USB standard specs are 5V regulated, 0.5A max draw.

Before plugging in the pi, the unit provides 5.1V.  After plugging in, the power draw is up to 0.6A and the voltage sags to 4.6-4.8V.  After plugging in the Pi, unplugging the Pi and plugging in the OEM thumb drive does not work.
Without power-Y cable.
[image]

With power-Y cable.
[image]

Now I can see the files on the drive via the app. 
[image]

Basically this is my setup
[image]

Using the adapter I mentioned above, I was able to reduce the voltage drop and prevent the power from being drained, thus resulting in the unit working

Parts mentioned: 
USB Power Splitter: https://www.amazon.com/gp/product/B07PV8ZN1X/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
Pi Zero W USB addon Board: https://www.amazon.com/MakerFocus-Raspberry-Required-Provide-Connector/dp/B077W69CD1/
USB Power Supply: https://www.amazon.com/Anker-Charger-PowerPort-iPhone-Galaxy/dp/B00P936188/
Pi Zero W: haha good luck.

adamoutler, 2022-02-15:
I've conducted several successful prints.  No issues. Closing.

closed 2022-02-15 by adamoutler